izhengfan/se2lam #5: Can't run robustly in pure rotating scene

LW73, 2019-07-30:
I have tested the program with self-collected data, and encountered several bugs.
1. In a pure rotating scene and the image is white wall, the framepub collapsed(don't refresh new frames), and it doesn't spawn new frame, with KF.id jumped to 32664 while it's supposed to be about 1000,  but the Track thread is still keep running.
2. I run the se2lam several times using the same rosbag, but the maps it draws differ a lot. Maybe it's because the calculation of rotation isn't very accurate.
I wonder how does it calculate the rotation with both the odometry and visual data, what's the mechanism it fuses those two ?

izhengfan, 2019-07-30:
Hello, can you provide some part of your data so that I can take a look?

LW73, 2019-08-01:
Sorry, I cannot provide the data, because it belongs to corporate secrets. But  I can describe the scene: the camera is mounted on a robot which wanders in a supermarket; there are lots of crossroads and the robot  turns 90 degrees first and then goes straight. 
I know this kind of motion is really challenging for visual SLAM, I tried to use optical flow method for tracking, but it's really difficult to adapt for local mapping(besides, the loop closing seems not work very well, it only draws red lines instead of refreshing published map.)
I have recorded the screen for a typical bug, how to send it to you?

izhengfan, 2019-08-02:
Well, if you can, pls send to zhengfann@foxmail.com

closed 2019-08-03 by LW73