izhengfan/se2lam #9: Dataset warehouse

Shashika007, 2019-08-20:
Hi @izhengfan , 

Great work! I read your both papers and tried your both repository including your ORB_SLAM2 odometry adaptation with your dataset attached (Dataset Room) in the repository. I wanted to give try with "Dataset warehouse" which you mentioned in your papers. The reason why i wanted to try is to see horizontal faced camera. I tried to create my own dataset using a camera and AMCL pose in "x,y,theta" , so far it did not give me a good results. It would be nice if you share "Dataset warehouse" with our community.  Well appreciated.

Thank you. 

Shashika

izhengfan, 2019-08-21:
Hi,  I am sorry Dataset Warehouse can not be shared because it comes from one of our partner  companies.

Shashika007, 2019-08-21:
Hi @izhengfan 

It is okay. Thank you. And one thing, are you using 'millimeters' instead of 'meters' as your odometry linear measurement units ? 

Thank you. 

Shashika

izhengfan, 2019-08-21:
yes, it is mm

Shashika007, 2019-08-21:
@izhengfan Thank you.

closed 2019-08-21 by Shashika007