izhengfan/se2lam #14: frame.id suddenly increases when testing on another dataset

pengpeng1997, 2019-12-19:
Hello there! I'm currently testing the repo on the OpenLORIS_Scene dataset and got one error. The frame.id increases suddenly and the system stops working but I can't figure out why. Here is the data in case you want to reproduce it: [](https://drive.google.com/open?id=1Zs7JI2wAuYid-79Krr_H-2nvv52zVq1b)
And the output about the error:
```
-- INFO TR: Add new KF at frame 1013
!! INFO MP: Prune KF 35
-- DEBUG LM: time 1.18576e-322, number of KF 4
-- DEBUG LM: time 1.30748 ms.
## DEBUG GM: loopTime = 1.27434, numKFs = 34, numMPs = 1006
-- INFO TR: Add new KF at frame 1027
-- DEBUG LM: time 6.91918e-310, number of KF 5
## DEBUG GM: loopTime = 1.30663, numKFs = -- DEBUG LM: time 2.4945 ms.
35, numMPs = 1016
-- INFO TR: Add new KF at frame 1037
-- DEBUG LM: time 6.91918e-310, number of KF 6
-- DEBUG LM: time 1.81965 ms.
## DEBUG GM: loopTime = 1.03737, numKFs = 36, numMPs = 1016
-- INFO TR: Add new KF at frame 1046
!! INFO MP: Prune KF 39
-- DEBUG LM: time 1.18576e-322, number of KF 6
-- DEBUG LM: time 3.25863 ms.
## DEBUG GM: loopTime = 2.36911, numKFs = 36, numMPs = 1016
-- INFO TR: Add new KF at frame 1062
## DEBUG GM: loopTime = 0.925425, numKFs = 37, numMPs = 1016
!! INFO MP: Prune KF 40
-- DEBUG LM: time 1.18576e-322, number of KF 6
-- DEBUG LM: time 2.15973 ms.
-- INFO TR: Add new KF at frame 32606
## DEBUG GM: loopTime = 0.783606, numKFs = 37, numMPs = 1016
-- DEBUG LM: time 6.91918e-310, number of KF 6
-- DEBUG LM: time 2.34234 ms.
```
Thanks if you have any idea about this problem!

izhengfan, 2019-12-20:
Thanks for the feedback, will look into it soon.

pengpeng1997, 2019-12-23:
Finally I found that the reason is none feature point can be extracted from one of the frame in my test data. So line 129 `mFrame = Frame(img, odo, mpORBextractor, Config::Kcam, Config::Dcam);` in ./src/Track.cpp directly `return`(line 29 in ./src/Frame.cpp) without assigning a value to the `id`. I think this kind of situation needs to be handled in a more robust way.

izhengfan, 2019-12-24:
Hi, good catch, I now put the lines initializing frame id before that return, check the latest update.

closed 2019-12-24 by pengpeng1997