izhengfan/se2lam #17: why I can't see the trajectory in RVIZ

HeadReaper-hc, 2020-03-12:
When I run your code and datasets, I can't see the trajectory. 

I set the variables `local_print, global_print, local_verbose, global_verbose` all to 1 in Setting.yaml for debugging, what I got in the terminal is as follows.

`###
###  se2lam: On-SE(2) Localization and Mapping with SE(2)-XYZ Constraints.
###

Loading ORB Vocabulary. This could take a while.
Vocabulary loaded!

# Load camera config ...
- Camera matrix: 
 [231.97603, 0, 326.92392;
 0, 232.15723, 227.83849;
 0, 0, 1]
- Camera distortion: 
 [-0.20740651;
 0.032194071;
 0.001120166;
 0.00085941155;
 0]
- Img size: 
 [640 x 480]

Running SLAM
-- INFO TR: Create first frame with 400 features.
## DEBUG GM: loopTime = 1.79287, numKFs = 1, numMPs = 0
Finish test...

# Finished. Saving keyframe trajectory ...
Exiting tracking ..
Exiting globalmapper ..
Exiting localmapper ..
Exiting mappublish ..
System is cleared ..
wait for finish finished...
Rec close...
Exit test...
`

I just want to know whether this is normal, thank you.

izhengfan, 2020-03-17:
The problem exits right after the 1st frame? Looks like data reading problem.

fbz1993, 2020-03-23:
i also encounter with the same problem, the output:
###
###  se2lam: On-SE(2) Localization and Mapping with SE(2)-XYZ Constraints.
###

Loading ORB Vocabulary. This could take a while.
Vocabulary loaded!

# Load camera config ...
- Camera matrix: 
 [231.97603, 0, 326.92392;
 0, 232.15723, 227.83849;
 0, 0, 1]
- Camera distortion: 
 [-0.20740651;
 0.032194071;
 0.001120166;
 0.00085941155;
 0]
- Img size: 
 [640 x 480]

Running SLAM
-- INFO TR: Create first frame with 400 features.
Finish test...

# Finished. Saving keyframe trajectory ...
Exiting tracking ..
Exiting globalmapper ..
Exiting localmapper ..
Exiting mappublish ..
System is cleared ..
wait for finish finished...
Rec close...
Exit test...
it sees that the program can run about one minute, than print"finish test....", but the trajectory can not be played in rviz, just two images imshowed.

fbz1993, 2020-03-23:
> i also encounter with the same problem, the output:
> ### se2lam: On-SE(2) Localization and Mapping with SE(2)-XYZ Constraints.
>
> Loading ORB Vocabulary. This could take a while.
> Vocabulary loaded!
> # [...]
the screenshot  of rviz:
[image]

fbz1993, 2020-03-23:
the screenshot of rviz
[image]

izhengfan, 2020-03-24:
Cannot figure out the problem with the given information.

fbz1993, 2020-03-24:
thank u for your quick response. i have another question about the robustness when the vehicle passes through the speed bump slowly(<10km/h).

fbz1993, 2020-03-25:
> When I run your code and datasets, I can't see the trajectory.
>
> I set the variables `local_print, global_print, local_verbose, global_verbose` all to 1 in Setting.yaml for debugging, what I got in [...]

please read the readme file,"To use either one of them, copy it to `odo_raw.txt` in `DatasetRoom/`",make sure the odo data can be read.

HeadReaper-hc, 2020-03-25:
@izhengfan @fbz1993 
thanks.It's work now.

izhengfan, 2020-03-28:
> thank u for your quick response. i have another question about the robustness when the vehicle passes through the speed bump slowly(<10km/h).

Can you describe the performance of such ill robustness?

fbz1993, 2020-03-28:
yes, I'm going to apply se2lam in autonomous parking field, but I'm not sure whether the algorithm can work in such scenes, especially out of SE(2) motion extremely. Can you give me some advice?

izhengfan, 2020-03-29:
I would suggest using SE2XYZ model if your field is planar with small perturbation, but not using the visual front-end. My front-end is not robust enough for actual use because it has not been extensively tested.

closed 2020-04-30 by izhengfan