izhengfan/se2lam #26: g2o EdgeSE3Expmap::linearizeOplus() Error?

Tofupi, 2020-06-12:
Hi, First thanks for your shared project named se2lam!!! Can you do me a favor:
In g2o source code, I read the EdgeSE3Expmap::linearizeOplus() function for odometry constrait edge. I inferred the jacobian matrix and get the result:
void EdgeSE3Expmap::linearizeOplus()
 {
VertexSE3Expmap * vi = static_cast<VertexSE3Expmap *>(_vertices[0]);
  SE3Quat Ti(vi->estimate());

  VertexSE3Expmap * vj = static_cast<VertexSE3Expmap *>(_vertices[1]);
  SE3Quat Tj(vj->estimate());

  const SE3Quat & Tij = _measurement;
  SE3Quat invTij = Tij.inverse();

//this is my result
  SE3Quat invTj = Tj.inverse();
  SE3Quat invTi_Tij = Tj.inverse()*Tij;
  _jacobianOplusXi = invTi_Tij.adj();
  _jacobianOplusXj = -invTj.adj();
}

But the souce code original result is:
void EdgeSE3Expmap::linearizeOplus() {
  VertexSE3Expmap * vi = static_cast<VertexSE3Expmap *>(_vertices[0]);
  SE3Quat Ti(vi->estimate());

  VertexSE3Expmap * vj = static_cast<VertexSE3Expmap *>(_vertices[1]);
  SE3Quat Tj(vj->estimate());

  const SE3Quat & Tij = _measurement;
  SE3Quat invTij = Tij.inverse();

//this is original result
  SE3Quat invTj_Tij = Tj.inverse()*Tij;
  SE3Quat infTi_invTij = Ti.inverse()*invTij;

   _jacobianOplusXi = invTj_Tij.adj();
  _jacobianOplusXj = -infTi_invTij.adj(); 
}

I modified original result to mine and compiled again. then run your project demo and it works well. I want to konw the right jacobian matrix. Can I have a little time to help me inferred it again can make sure if I am right or not. Thank you very much!!!

Tofupi, 2020-06-12:
[image]
I inferred as above

izhengfan, 2020-06-14:
I don't understand your logic, but here is how g2o works:

[image]

Tofupi, 2020-06-15:
Thanks for your patience. I misunderstand the inferring way. Thanks again

closed 2020-06-15 by Tofupi