izhengfan/se2lam #29: Bug with code?

cy-goh, 2020-08-03:
Hello, 

in Track.cpp,

for line 361, Shouldn't  `bool c5 = fdOdo.theta >= 0.0349f ` be `bool c5 = fabs(dOdo.theta) >= 0.0349f`?

And for line 364, after converting to camera frame CTC, shouldnt it be `cv::Mat xy =  cTc.rowRange(0,3).col(3)`?

```
        cv::Mat xy = cTc.rowRange(0,2).col(3);
        bool c6 = cv::norm(xy) >= ( 0.0523f * Config::UPPER_DEPTH * 0.1f ); // 3 degree = 0.0523 rad
```

CrisGao, 2020-08-18:
When the code for checking parallax is changed to `return cosParallax >= minCos[minDegree-1];`,then there are few feature points that can be successfully triangulated, and the calculation results basically rely on the data of odom

cy-goh, 2020-08-18:
@CrisGao , yes you are right. I made a mistake on the cosine.

izhengfan, 2020-08-18:
@cy-goh Good catch.

@CrisGao Thanks for the comment. According to your comment, I guess the current `checkParallax()` does not need revision?

CrisGao, 2020-08-19:
@izhengfan ,yes `checkParallax()`is not wrong.If I understand correctly, I think `cv::Mat xy = cTc.rowRange(0,2).col(3);` is not wrong, cv::norm(xy) is the Euclidean distance of the plane

izhengfan, 2020-08-19:
> @izhengfan ,yes `checkParallax()`is not wrong.If I understand correctly, I think `cv::Mat xy = cTc.rowRange(0,2).col(3);` is not wrong, cv::norm(xy) is the Euclidean distance of the plane

Thanks for the reminding, let me rethink it.

cy-goh, 2020-08-19:
> @izhengfan ,yes `checkParallax()`is not wrong.If I understand correctly, I think `cv::Mat xy = cTc.rowRange(0,2).col(3);` is not wrong, cv::norm(xy) is the Euclidean distance of the plane

I think cTc is in the camera coordinate frame, with y axis pointing down.

izhengfan, 2020-08-19:
> > @izhengfan ,yes `checkParallax()`is not wrong.If I understand correctly, I think `cv::Mat xy = cTc.rowRange(0,2).col(3);` is not wrong, cv::norm(xy) is the Euclidean distance of the plane
> 
> I think cTc is in the camera coordinate frame, with y axis pointing down.

Yes, the physical meaning of `c6` is whether the distance between the camera center in two moments is large enough to get enough parallax, so it should be like `xy = cTc.rowRange(0,3).col(3)`.

CrisGao, 2020-08-19:
Sorry, I'm still a bit confused. `0.0523f * Config::UPPER_DEPTH * 0.1f` is equivalent to an arc length with a radius of 1m and an angle of 3 degrees. Compared with `cTc.rowRange(0,2).col(3)`, `cTc.rowRange(0,3).col(3)`considers the Z value. I just think cTc should be a translation on the plane.Sorry to disturb you

CrisGao, 2020-08-19:
Sorry, I'm still a bit confused. `0.0523f * Config::UPPER_DEPTH * 0.1f` is equivalent to an arc length with a radius of 1m and an angle of 3 degrees. Compared with `cTc.rowRange(0,2).col(3),` `cTc.rowRange(0,3).col(3)` considers the Z value. I just think cTc should be a translation on a plane.Sorry to bother you

CrisGao, 2020-08-19:
I’m sorry that I misunderstood it. Only when the camera is facing up, `cTc.rowRange(0,2).col(3)` is the translation of the xy plane.

izhengfan, 2020-08-19:
> I’m sorry that I misunderstood it. Only when the camera is facing up, `cTc.rowRange(0,2).col(3)` is the translation of the xy plane.

Yes, when the camera is facing forward or to the side, the z component in `cTc` can affect the parallax.

closed 2020-08-19 by izhengfan

cy-goh, 2020-08-19:
@izhengfan i found quite a fair bit of other bugs. Do you want me to open a new issue, or continue here?

izhengfan, 2020-08-19:
> @izhengfan i found quite a fair bit of other bugs. Do you want me to open a new issue, or continue here?

Actually you can fork and open a PR. Thus your contributions will be counted.