izhengfan/se2lam #30: Why does Jacobian calculation only use fx and not fy in optimization

CrisGao, 2020-08-18:
I found that the Jacobian calculations in the program only involve fx, such as SE(2)-XYZ constrained Jacobian calculations:
**src/EdgeSE2XYZ.cpp/linearizeOplus()**
[image]
and **Thirdparty/g2o/g2o/types/sba/types_six_dof_expmap.cpp/linearizeOplus()**
[image]

izhengfan, 2020-08-18:
We use only fx because in the early-stage we use the types of g2o, and `g2o::CameraParameters` only has one parameter `focal_length`, i.e. they consider `fx==fy`. This is correct in all the sensors we met with, so we just keep consistent to it.

CrisGao, 2020-08-19:
I got it,Thank you

closed 2020-08-19 by izhengfan