izhengfan/se2lam #36: 您好，我编译您的代码出现些错误，错误显示找不到g2o定义的结构，但是我查看电脑由存在这些结构

leigetiwen, 2020-12-25:
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:63:45: error: ‘Vector3D’ was not declared in this scope
 class PreEdgeSE2 : public BaseBinaryEdge<3, Vector3D, VertexSE2, VertexSE2>
                                             ^~~~~~~~
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:63:45: note: suggested alternative: ‘Vector3’
 class PreEdgeSE2 : public BaseBinaryEdge<3, Vector3D, VertexSE2, VertexSE2>
                                             ^~~~~~~~
                                             Vector3
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:63:75: error: template argument 2 is invalid
 class PreEdgeSE2 : public BaseBinaryEdge<3, Vector3D, VertexSE2, VertexSE2>
                                                                           ^
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h: In member function ‘void g2o::PreEdgeSE2::computeError()’:
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:71:61: error: ‘_vertices’ was not declared in this scope
         const VertexSE2* v1 = static_cast<const VertexSE2*>(_vertices[0]);
                                                             ^~~~~~~~~
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:73:9: error: ‘Matrix2D’ was not declared in this scope
         Matrix2D Ri = v1->estimate().rotation().toRotationMatrix();
         ^~~~~~~~
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:73:9: note: suggested alternative: ‘Matrix2’
         Matrix2D Ri = v1->estimate().rotation().toRotationMatrix();
         ^~~~~~~~
         Matrix2
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:74:9: error: ‘Vector2D’ was not declared in this scope
         Vector2D ri = v1->estimate().translation();
         ^~~~~~~~
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:74:9: note: suggested alternative: ‘Vector2’
         Vector2D ri = v1->estimate().translation();
         ^~~~~~~~
         Vector2
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:77:18: error: expected ‘;’ before ‘rj’
         Vector2D rj = v2->estimate().translation();
                  ^~
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:79:9: error: ‘_error’ was not declared in this scope
         _error.head<2>() = Ri.transpose() * (rj-ri) - _measurement.head<2>();
         ^~~~~~
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:79:9: note: suggested alternative: ‘perror’
         _error.head<2>() = Ri.transpose() * (rj-ri) - _measurement.head<2>();
         ^~~~~~
         perror
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:79:24: error: expected primary-expression before ‘)’ token
         _error.head<2>() = Ri.transpose() * (rj-ri) - _measurement.head<2>();
                        ^
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:79:28: error: ‘Ri’ was not declared in this scope
         _error.head<2>() = Ri.transpose() * (rj-ri) - _measurement.head<2>();
                            ^~
/home/jetson/simulation_ws/src/se2lam/include/se2lam/EdgeSE2XYZ.h:79:28: note: suggested alternative: ‘ai’
         _error.head<2>() = Ri.transpose() * (rj-ri) - _measurement.head<2>();

izhengfan, 2020-12-26:
找不到 Vector3D，请确保你用的 g2o 版本和我一样

leigetiwen, 2020-12-28:
找到问题了，是由于电脑安装了多个g2o

closed 2020-12-28 by leigetiwen